eclipse-threadx/threadx #1: Use clang-format to ensure consistent style?

jserv, 2020-05-20:
Several Microsoft open source projects already use [clang-format](https://clang.llvm.org/docs/ClangFormat.html) to indent, such as [ONNX Runtime](https://github.com/microsoft/onnxruntime) and [STL](https://github.com/microsoft/STL). Will ThreadX project take it for consistent code style?

jserv, 2020-05-20:
Take [samples/demo_threadx.c](https://github.com/azure-rtos/threadx/blob/master/samples/demo_threadx.c#L71) for example. The style was inconsistent.

PProvost, 2020-05-21:
That's a good idea. We will investigate and update this issue when we've got an answer.

PProvost, 2020-07-31:
We will continue to investigate this but can't commit to it at this point. Closing the issue for now.

closed 2020-07-31 by PProvost