eclipse-threadx/threadx #3: RasPi support?

BrianCatlin, 2020-05-20:
When will the Raspberry Pi be supported?

BillLamie, 2020-05-27:
Currently, we do not have Azure RTOS support for Raspberry Pi on our near term roadmap. That said, it would be good to get more information on what your particular use case looks like and which version of Raspberry Pi hardware you are interested in. With your additional information we can further discuss on our side.

BrianCatlin, 2020-05-28:
The RasPi is cheap and ubiquitous, and as such it is the perfect platform for experimentation and development, both from the hardware and software perspective. Spending $40 for a RasPi for each software developer to use while a hardware platform is being developed makes a lot of sense (and cents). It would be great if you supported both the 3 and the 4, but I think that most people could live with the 4. Supporting the RasPi would dramatically increase the number of developers that would try azure-rtos

BillLamie, 2020-05-28:
Thanks for the additional input Brian, this is definitely on our radar!

PProvost, 2020-05-29:
I'm going to close this now. Feel free to re-open if you have additional context or information.

JinShil, 2020-06-19:
I think it would be nice good to keep this issue open to collection information from other interested parties.  The Raspberry Pi Zero (being only $5) makes it a very attractive alternative to the typical ARM Cortex-M devices.

The Raspberry Pi Foundation also creates a Compute Module that is designed to be embedded in custom industrial devices.

If all of the various Raspberry Pi features were functional under the Azure RTOS, I would definitely be interested in designing products with it.  Some of the products in mind are panel PCs, PLCs, and even PC peripherals (e.g. keyboards and other human interface devices).

I am, personally and professionally, invested in the Raspberry Pi ecosystem, and having the ability to boot the hardware in a few seconds with a small RTOS would be game changer for some of the products I design and work with.

goldscott, 2020-06-25:
I'll reopen the issue. We do have ThreadX available for the ARMv8-A family, so theoretically, it will run on the RasPi. It's the BSP that may need some work...

JinShil, 2020-06-25:
According to [this post on the Raspberry Pi Forum](https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=255314&p=1558243&hilit=ThreadX) ThreadX is already used to boot the core.  Does that also include a ARMv6Z port for the Raspberry Pi Zero?

goldscott, 2020-06-25:
I'm not sure where that info comes from - we have never done any work with the RasPi. 
We have ThreadX for ARM11, though I think we are deprecating it because the ARM11 is so old.

JinShil, 2020-06-25:
> I'm not sure where that info comes from - we have never done any work with the RasPi.

I suspect it was something the folks at the Raspberry Pi Foundation, Broadcom, and Express Logic worked on.  The posts in that thread are directly from the folks at the Raspberry Pi Foundation.

> We have ThreadX for ARM11, though I think we are deprecating it because the ARM11 is so old.

If you do choose to deprecate it, I hope you'll consider open-sourcing it for the broader community to potentially fork and maintain.  That would be my request for any port that you have but no longer intend to maintain.

goldscott, 2020-06-26:
I've been with Express Logic and Microsoft for 7 years now, and Broadcom has used ThreadX, but I don't recall ever interacting with the R Pi folks. I'm asking internally about releasing the ARM11. Stay tuned.

goldscott, 2020-06-30:
@JinShil We'll likely put the ARM11 port up on github.

alantsd, 2020-07-07:
Please make azure-RTOS support the RasPi, a thousand thanks!

JakeSays, 2020-07-12:
The problem with azure-rtos on the RPI is licensing. Azure-rtos isn't open source, so the costs of using it on an rpi reduce the appeal greatly.

alantsd, 2020-07-12:
Please explain more, thank you.

JakeSays, 2020-07-12:
@alantsd from license.txt:

> General. You may install and use the software and the included Microsoft
applications solely for internal development, testing and evaluation purposes.
Any distribution or production use requires a separate license as set forth in
Section 2.

That separate license is a commercial license. If your product uses hardware from a hardware partner then (again see license.txt) you are covered by that hardware partner's license with Microsoft. Otherwise a commercial license must be purchased. To date I am only aware of one hardware partner: Microsoft. If you use an azure sphere in your product then you can use azure-rtos.

I have read elsewhere, though, that there are agreements in the works with other hardware vendors (microchip, etc).

alantsd, 2020-07-13:
Thank you for the information on license.
That would likely means PI community will never get azure rtos official support.

goldscott, 2020-07-14:
Like @JakeSays quoted, ThreadX is free to develop with, it's distributing and producing a commercial product that requires a license, so you can certainly work on getting ThreadX running on the Pi. We're focused on getting ThreadX running on generic architectures (e.g. Cortex-A53), not necessarily specific hardware platforms (e.g. Raspberry Pi).
There seems to be lots of interest from the community to get ThreadX running on the Pi, so please work on it together!

sukesh-ak, 2020-07-21:
**Raspberry Pi Zero W** would be a good option for people like me to get started with Azure RTOS. Transition from prototype to product becomes easier too without much investment in the beginning. The EVK's listed over here are all expensive ones to tinker or get started with.

closed 2020-07-28 by PProvost

RushOnline, 2021-02-16:
According to https://en.wikipedia.org/wiki/VideoCore RPI already managed by ThreadX. So what effect you expect if you can boot VPU with ThreadX? Effect will be nothing because hardware (VC4) is proprietary (Broadcom).

goldscott, 2021-02-16:
You'll have to contact the VideoCore and RPI folks - Express Logic/Microsoft did not do any work on the RPI.