eclipse-threadx/threadx #6: Plan for SMP support?

hungys, 2020-05-24:
As I know, ThreadX had been added SMP support long time ago, but it seems that multicore SMP feature was not included in Azure RTOS release, any plan on this?

yuxin-azrtos, 2020-05-26:
We will SMP version released on Github soon. Is there a particular processor/toolchain you are looking for?

mfkiwl, 2020-05-26:
How about Rockchip's RK3308? It has 4 cortex A-35 cores and doesn't have GPU, so easier to port.

yuxin-azrtos, 2020-05-29:
@mfkiwl We do plan to support A-35 SMP using ARM and GCC toolchain.  It will be available on GH later this year.

closed 2020-05-29 by yuxin-azrtos

mfkiwl, 2020-05-29:
> @mfkiwl We do plan to support A-35 SMP using ARM and GCC toolchain. It will be available on GH later this year.

Good news for me. Can't wait to see it happen!

hungys, 2020-05-31:
@yuxin-azrtos Any plan for MIPS or RISC-V processor support? Thanks.

mycan12345, 2021-08-16:
> How about Rockchip's RK3308? It has 4 cortex A-35 cores and doesn't have GPU, so easier to port.

Hi, have you port threadx to rk3308?

goldscott, 2021-08-16:
@mycan12345 A35 is here: https://github.com/azure-rtos/threadx/tree/master/ports_smp/cortex_a35_smp