eclipse-threadx/threadx #12: Win32 or Linux Port?

edbek, 2020-06-25:
ThreadX requires Win32 or Linux simulation ! Like for example FreeRTOS or uCOS-III.
Can we expect such an implementation for the future ?

goldscott, 2020-06-25:
I'm not sure I understand your question. We have win32 and linux ports for ThreadX - these will be added to github soon.

edbek, 2020-06-26:
Yes, this is the win32 and linux port.
Thank.

goldscott, 2020-06-26:
The win32 port should be added very soon (next week or two) and linux will come after. Closing.

closed 2020-06-26 by goldscott

goldscott, 2020-07-10:
We are about to publish a bunch of new ports, including win32 and linux, either today or early next week.

edbek, 2020-07-11:
We wait. Thank you !

goldscott, 2020-07-30:
Windows and Linux ports were posted. https://github.com/azure-rtos/threadx/tree/master/ports

edbek, 2020-07-30:
Yes, we have seen and are testing. Thank you.