eclipse-threadx/threadx #13: Require documentation for undefined reference to `_tx_initialize_low_level'

coder137, 2020-06-26:
## EXPLANATION

1. I compiled threadx in a different folder using the documentation 
```
$ cmake -Bbuild -DCMAKE_TOOLCHAIN_FILE=cmake/cortex_m4.cmake -GNinja .
$ cmake --build ./build
```
2. Proceeded to add `tx_api.h`, `tx_port.h` and `libthreadx.a` to the threadx folder of my Minimal STM32 Project

3. Updated CMake like so
```cmake
# Add Threadx
target_include_directories(${USER_PROJECT_TARGET} PRIVATE threadx)
target_link_directories(${USER_PROJECT_TARGET} PRIVATE threadx)
target_link_libraries(${USER_PROJECT_TARGET} PRIVATE threadx)
```

4. CMake generates the project successfully

5. Updated main.c

```cpp
TX_THREAD my_thread;
void my_thread_entry(ULONG thread_input) {
  while (1) {
    uart_interrupt__write_string(&uart_interrupt_config,
                                 "Hello from Interrupt\r\n");
    tx_thread_sleep(100);
  }
}

void tx_application_define(void *first_unused_memory) {

  /* Create my_thread! */
  tx_thread_create(&my_thread, "My Thread", my_thread_entry, 0x1234,
                   first_unused_memory, 1024, 3, 3, TX_NO_TIME_SLICE,
                   TX_AUTO_START);
}

int main(void) {

  // UART Config
  main__uart_init();
  main__uart_interrupt_init();

  // uart__write_string(&uart_config, "Hello World\r\n");
  // _spin_delay(1000 * 1000);
  uart_interrupt__write_string(&uart_interrupt_config,
                               "Hello from Interrupt\r\n");

  tx_kernel_enter();

  while (1) {
  }

  return 0;
}
```

5. **ERROR** when running
```
cmake --build build
```
**D:\Code\threadx\threadx\build/../common/src/tx_initialize_kernel_enter.c:117: undefined reference to `_tx_initialize_low_level'**

I took a look at the above-mentioned function used in this example [Getting Started STM32L4 Example](https://github.com/azure-rtos/getting-started/blob/master/STMicroelectronics/STM32L4/app/startup/tx_initialize_low_level.S)

```asm
@VOID   _tx_initialize_low_level(VOID)
@{
    .global  _tx_initialize_low_level
    .thumb_func
_tx_initialize_low_level:
@
@    /* Disable interrupts during ThreadX initialization.  */
@
    CPSID   i
@
@    /* Set base of available memory to end of non-initialised RAM area.  */
@     
    LDR     r0, =_tx_initialize_unused_memory       @ Build address of unused memory pointer
    LDR     r1, =__RAM_segment_used_end__           @ Build first free address   
    ADD     r1, r1, #4                              @ 
    STR     r1, [r0]                                @ Setup first unused memory pointer
@
@    /* Setup Vector Table Offset Register.  */
@    
    MOV     r0, #0xE000E000                         @ Build address of NVIC registers
    LDR     r1, =g_pfnVectors                       @ Pickup address of vector table  
    STR     r1, [r0, #0xD08]                        @ Set vector table address   
@
@    /* Set system stack pointer from vector value.  */
@
    LDR     r0, =_tx_thread_system_stack_ptr        @ Build address of system stack pointer
    LDR     r1, =g_pfnVectors                       @ Pickup address of vector table
    LDR     r1, [r1]                                @ Pickup reset stack pointer
    STR     r1, [r0]                                @ Save system stack pointer
@
@    /* Enable the cycle count register.  */
@
    LDR     r0, =0xE0001000                         @ Build address of DWT register
    LDR     r1, [r0]                                @ Pickup the current value
    ORR     r1, r1, #1                              @ Set the CYCCNTENA bit
    STR     r1, [r0]                                @ Enable the cycle count register 
@
@    /* Configure SysTick for 100Hz clock, or 16384 cycles if no reference.  */
@
    MOV     r0, #0xE000E000                         @ Build address of NVIC registers
    LDR     r1, =SYSTICK_CYCLES
    STR     r1, [r0, #0x14]                         @ Setup SysTick Reload Value
    MOV     r1, #0x7                                @ Build SysTick Control Enable Value
    STR     r1, [r0, #0x10]                         @ Setup SysTick Control

@
@    /* Configure handler priorities.  */
@
    LDR     r1, =0x00000000                         @ Rsrv, UsgF, BusF, MemM
    STR     r1, [r0, #0xD18]                        @ Setup System Handlers 4-7 Priority Registers

    LDR     r1, =0xFF000000                         @ SVCl, Rsrv, Rsrv, Rsrv
    STR     r1, [r0, #0xD1C]                        @ Setup System Handlers 8-11 Priority Registers
                                                    @ Note: SVC must be lowest priority, which is 0xFF

    LDR     r1, =0x40FF0000                         @ SysT, PnSV, Rsrv, DbgM
    STR     r1, [r0, #0xD20]                        @ Setup System Handlers 12-15 Priority Registers
                                                    @ Note: PnSV must be lowest priority, which is 0xFF
        
@
@    /* Return to caller.  */
@    
    BX      lr 
@}
@
```

## REQUIREMENT

If possible I would like to get an example for the above function in **C** since my startup code and interrupt vector table (entire project) has been written in C. [Link to my Minimal Projects for STM32](https://github.com/coder137/STM32-Repo/tree/master/Template)

If possible I would also like to know if there is more documentation to learn about the internal working of Threadx.

goldscott, 2020-06-26:
We do not have this file in C, only assembly. 

There is the user guide here: https://docs.microsoft.com/en-us/azure/rtos/threadx/about-this-guide
And a book available for purchase: https://www.amazon.com/Real-Time-Embedded-Multithreading-Using-ThreadX/dp/1523730420

coder137, 2020-06-27:
Would there be any roadmap for a C based version?

Also I went through the documentation and I was unable to find any reference for `_tx_low_level_initialization` usage and implementation.
Please point me in the right direction for this as well! Thank you

goldscott, 2020-06-30:
There is no roadmap for a C version as of now. This file is where the SysTick is configured. Doesn't look like we have our port-specific readmes up yet, but we'll be putting them up in the next few days.

coder137, 2020-07-01:
Please update with the link here as well when the READMEs are live.
That should resolve this issue.

goldscott, 2020-07-14:
Hi @coder137 - the readme_threadx.txt files are now live. See here for example: https://github.com/azure-rtos/threadx/tree/master/ports/cortex_m4/iar
The readme files for the Cortex-M0/M3/M4/M7 GCC ports did not make it up for some reason. I'll get them up ASAP. 
We've added a bunch of new processors and toolchains. More coming soon.

closed 2020-07-14 by goldscott

coder137, 2020-07-14:
Thank you for your help @goldscott !

reynoldsbd, 2021-04-22:
@goldscott would it be possible to update Chapter 2 of the user guide with some conceptual information about `_tx_low_level_initialization`?

I can certainly adapt the one from *ports/.../example_build/*, but it would be useful to know - from the horse's mouth, so to speak - which parts are important vs. which are specific to the sample application.

goldscott, 2021-04-22:
Hello @reynoldsbd - we try to keep the port-specific stuff out of the user guide, but I agree with you - some more info about what this function does would be helpful. It's on my to-do list to add more detail in the user guide. 

This function really has only one requirement: it must set up the _tx_initialize_unused_memory global with the first free memory address. We also usually set up the timer interrupt here as well, but that is port-specific. For something like the Cortex-M, it's easy to set up the SysTick here (and users can configure the timer by modifying the #defines at the top of the file), but for other architectures like Cortex-A, it's up to the user to configure a timer elsewhere.

coder137, 2021-04-22:
@reynoldsbd I managed to convert the `_tx_low_level_initialization` from assembly to C

Please reference this issue https://github.com/azure-rtos/threadx/issues/60

The C code for tx_low level is here. The code has very verbose comments and I hope it helps. (https://github.com/coder137/STM32-Repo/blob/master/Template/Minimal_Threadx/application/tx_initialize_low_level.c)

The assembly equivalent is here (https://github.com/coder137/STM32-Repo/blob/master/Template/Minimal_Threadx/application/tx_initialize_low_level.S)