eclipse-threadx/threadx #14: Licensed Hardware list

josesimoes, 2020-07-03:
I can see on the Azure website this list of hardware partners: https://azure.microsoft.com/en-us/services/rtos/#partners
 which I presume are (will be?) the so called "pre-licensed devices". Is this correct? If yes, can the list on `LICENSED-HARDWARE.txt` be updated to reflect this?

Thanks!

goldscott, 2020-07-03:
Hi @josesimoes - we'll be updating this list once the deals with each semi manufacturer are finalized. We're aiming for Microchip, NXP, Renesas, ST, and Qualcomm. Should be very soon!

josesimoes, 2020-07-03:
Good to know.
Knock at TI and Nordic door too... 😉

LP-33, 2020-07-14:
Hello,

having ThreadX and the middleware open source is great, really.
But without knowing which hardware we can use for using it freely is a real problem.

We are starting a new project based on  Kinetis K64/66 and we need to know if this MCU will be part of the pre-licensed devices.

Do you know when you could provide a preliminary list ?

Thank you.

goldscott, 2020-07-14:
Hi @LP-33 - I feel your pain! These giant corporations and big deals seem to move very slowly. 

The Kinetis K64/66 family should be on the list once these deals are closed, but I don't know when that will happen.

LP-33, 2020-07-15:
Hi goldscott,

ok, this is good news.
Did you plan to provide NetX/NetX-Duo ports for these MCU before that ? As you did for iMXRT1060, SAME54, STM32F746 and STM32L475?
Thank you.

goldscott, 2020-07-15:
Hi @LP-33 - NetX and NetX Duo are available for the Cortex-M series, so they will work on the K64/66. We will not be making sample projects for every processor/dev board out there. NXP (and our other partners, ST, Microchip, etc) will likely make sample projects for their particular processors and dev boards.

closed 2020-07-31 by PProvost

Polarisru, 2020-08-17:
Hello, how about ATSAMC21 MCUs (Cortex-M0, Microchip), will they be available for free? We are working at some new projects right now and have used FreeRTOS there, but it is still possible to change the RTOS, so we are very interested to get Azure RTOS working because of MISRA compatibility.

JakeSays, 2020-08-17:
@Polarisru I'm a microchip customer and I was lead to believe that azure rtos is available at no cost for microchip controllers. I never pursued it though because we are also using an azure sphere which gives us license to use azure rtos on any mcu associated with the project.

goldscott, 2020-08-17:
Hi @Polarisru - using Azure RTOS on Cortex-M devices from Microchip should be free, but we have not yet closed that deal. You are allowed to begin your development for free, so you can get started with ThreadX today.

LP-33, 2020-12-07:
Hello,
did you plan to add a multi-core Cortex Ax(x) in the licensed list? (e.g. iMX8M Nano).
Thank you.

goldscott, 2020-12-07:
@LP-33 - you'll have to contact the manufacturers that make those processors and ask (e.g. NXP, Renesas, etc). They choose what gets put on the list.

Taranus, 2021-12-08:
Hello,

do you know if it is planned to add the Raspberry Pi Pico to this list?

Thank you.

goldscott, 2021-12-08:
@Taranus - we have no plans to add RPi to the licensed hardware list.

pnchatterji, 2022-11-24:
Hi Goldscott,

is there any possibility of using Azur RTOS on Nordic hardware in future? Is it in the works?

Otherwise, is there a possibility of using just the FileX component on top of Zephyr RTOs, for example, which is supported by Nordic? I understand FileX has a Standalone mode, so it should be possible technically. How does it look from a legal standpoint? Does one have to obtain a special license from Microsoft for this?
thanks
regards
Pashupatu

goldscott, 2022-12-20:
@pnchatterji - we do not have any agreement with Nordic, so you would need an Azure RTOS license. 

Technically possible, though you would need to license Azure RTOS to use FileX.