eclipse-threadx/threadx #15: why enable the irq so early on pensv handler of M4 port?

13824125580, 2020-07-05:
HI guys:

   i found there is somthing differnce withe pendsv critical section install compare to the ucos and zephyr port on Cortex-M
on ucos, all the pendsv are a big critical setions areas so the thread switch op cant be interrupted by async things.

but on threadx porting, it seems the "CPSIE I"  was executed before the" __tx_ts_restore " of exectuing, it this safe if 
interrupt happend just after flow of cpsie i or some instructions later of folllow ing instruct series?

thank you!
__tx_ts_new:
@      
@    /* Now we are looking for a new thread to execute!  */
@      
    CPSID   i                                       @ Disable interrupts      
    LDR     r1, [r2]                                @ Is there another thread ready to execute?
    CBZ     r1, __tx_ts_wait                        @ No, skip to the wait processing
@      
@    /* Yes, another thread is ready for else, make the current thread the new thread.  */
@      
    STR     r1, [r0]                                @ Setup the current thread pointer to the new thread 
    **CPSIE   i                                       @ Enable interrupts**       
@      
@    /* Increment the thread run count.  */
@      
__tx_ts_restore:
    LDR     r7, [r1, #4]                            @ Pickup the current thread run count
    LDR     r4, =_tx_timer_time_slice               @ Build address of time-slice variable
    LDR     r5, [r1, #24]                           @ Pickup thread's current time-slice
    ADD     r7, r7, #1                              @ Increment the thread run count                                                                                                                        
    STR     r7, [r1, #4]                            @ Store the new run count
@      
@    /* Setup global time-slice with thread's current time-slice.  */

goldscott, 2020-07-07:
Yeah, ThreadX is designed to run as fast as possible and minimize critical sections. Are you observing any problems in the code here?

13824125580, 2020-07-08:
not yet, just feel it is not like others rtos.

so, can you show me how to run threadx on a M4 boards? how the software and boards can should i prepare?
and how to genernate the execuable elf file by threadx sdk because now i can only get libthread.a by cmake?
any guidlines?
thank you!

13824125580, 2020-07-08:
@goldscott 
thank you!
not yet, just feel it is not like others rtos.

so, can you show me how to run threadx on a M4 boards? how the software and boards can should i prepare?
and how to genernate the execuable elf file by threadx sdk because now i can only get libthread.a by cmake?
any guidlines?
thank you!

goldscott, 2020-07-10:
Import all the ThreadX code into your IDE (or create your own makefile), build ThreadX as a library, and then build the demo_threadx.c file in the samples directory as your application. See https://github.com/azure-rtos/samples for a bunch of samples in various IDEs.

goldscott, 2020-07-14:
We've just added a bunch of new processors and toolchain support, with examples.