eclipse-threadx/threadx #17: no member named 'bsd errno' in 'struct TX_THREAD_STRUCT'

nstcl, 2020-07-14:
addons\BSD\nxd_bsd.c(9562): error: no member named 'bsd_errno' in 'struct TX_THREAD_STRUCT'

bo-ms, 2020-07-14:
You need to do following setting for building the ThreadX and NetX Duo components of a BSD application.
 
**ThreadX** 

The ThreadX library must define bsd_errno in the thread local storage.  We recommend the following procedure:

1.	In tx_port.h, set one of the TX_THREAD_EXTENSION macros as follows:

        #define TX_THREAD_EXTENSION_3       int bsd_errno                       

2.	Rebuild the ThreadX library. 

Note that if TX_THREAD_EXTENSION_3 is already used, the user is free to use one of the other TX_THREAD_EXTENSION macros.

**NetX Duo** 

Before using NetX Duo BSD Services, the NetX Duo library must be built with NX_ENABLE_EXTENDED_NOTIFY_SUPPORT defined.  By default it is not defined.  If the BSD raw sockets are to be used, the NetX Duo library must be built with NX_ENABLE_IP_RAW_PACKET_FILTER defined.

ameya7295, 2020-07-14:
Hi @bo-ms 

I did add a int bsd_errno in tx_api.h
Also, my question why can't we name it as tx_errno in tx_api.h, and in the nx_bsd.c and nxd_bsd.c we can have a bsd_errno while setting it ie.
```
VOID set_errno(INT bsd_errno)
{
...
current_thread_ptr -> tx_errno = bsd_errno; 
...
}
```
Just a suggestion,

goldscott, 2020-07-14:
@ameya7295 why do you suggest the name tx_errno rather than bsd_errno?

ameya7295, 2020-07-15:
@goldscott Yes,I am suggesting the same because, if in future we want to use this variable in some other way then we will have to set the bsd_errno which is in tx_api.h . Which will be quite confusing to understand which errno to set. As ThreadX follows the noun verb naming convection user will will be searching for tx_errno.
What is your take on this ?

goldscott, 2020-07-21:
Hi @ameya7295 - we're going to continue using bsd_errno in the TX_THREAD struct. The member bsd_errno is only needed when using our BSD layer.

closed 2020-07-21 by goldscott

rubenax97, 2024-06-19:
You should add `;` at the final of define.
`#define TX_THREAD_EXTENSION_3       int bsd_errno;`