eclipse-threadx/threadx #32: RH850 port

tedacy, 2020-09-08:
Do you plan to port to RH850? I see that ThreadX has passed the ASIL D certification of ISO26262.

goldscott, 2020-09-08:
Hello @tedacy - we do have V850 and V850E ports, but they are not yet on github. 

Correct, ThreadX is ASIL D and 26262 certified.

tedacy, 2020-09-09:
> 
> 
> Hello @tedacy - we do have V850 and V850E ports, but they are not yet on github.
> 
> Correct, ThreadX is ASIL D and 26262 certified.

Thank you very much for your reply. Could you please provide me with V850 and V850E ports first, my email is chaoyang.ren.aj@renesas.com

goldscott, 2020-09-09:
Emailed link. Closing.

closed 2020-09-09 by goldscott

mikisama, 2020-12-22:
> 
> 
> Emailed link. Closing.

Hi @goldscott, I am interested in V850 ports too. Would you mind provide me too? My email is zhengtao404@qq.com

goldscott, 2021-01-20:
Hi @mikisama - I just emailed you a link.

daviddai817, 2021-10-31:
Hi, @goldscott. Can you also mail me the port for RH850? Email is 45201258@qq.com. Thank you。

goldscott, 2021-11-01:
@daviddai817 I emailed a link.

zwjkim, 2022-02-17:
> Hello @tedacy - we do have V850 and V850E ports, but they are not yet on github.
> 
> Correct, ThreadX is ASIL D and 26262 certified.

I am interested threadx and V850/RH850 ports. Could you provide me the ports, my email is zwjkim@gmail.com

kvvipa0028, 2023-05-20:
Hi .  I am interested threadx and V850/RH850 ports. Could you send to me , my email is  125130744@qq.com. I hope to try this .

TiejunMS, 2023-05-23:
Shared V850 port through OneDrive. Please check.

trinhdoanh, 2024-08-15:
Hi, 

Could anyone share me V850/RH850 porting too?
my email: tri.nguyen.wh@renesas.com or trongtribk06@gmail.com.

Thank you in advance!

trinhdoanh, 2024-08-19:
Hi @TiejunMS @goldscott,

Would you please share me RH850/V850 porting? My email trongtribk06@gmail.com

Thank you very much in advance!

Best Regards,
Tri Nguyen

Ban-3300, 2024-09-26:
Hello, is ThreadX for V850 still available? Could you mail me a ports for V850 and V850ES?
My email is andrija_ban@yahoo.com
Thanks

aholtzma-am, 2025-02-14:
We are also looking for v850. Any reason it isn't on GitHub yet?

ruinned, 2025-04-01:
I would like to have the link or OneDrive share as well, would you please send me one? Thanks. @goldscott @TiejunMS 
My email is sunericyl@outlook.com