eclipse-threadx/threadx #39: region `UNPLACED_SECTIONS' overflowed by 4 bytes

sanbrother, 2020-09-24:
I'm trying to run ThreadX on a stm32f0 (cortex-m0) chip. And the final step, an error occurs.
`
region 'UNPLACED_SECTIONS' overflowed by 4 bytes
`

Here is the sample LD file from official code, why there is something related to CM3???
```
MEMORY
{
  UNPLACED_SECTIONS (wx) : ORIGIN = 0x100000000, LENGTH = 0  
  CM3_System_Control_Space (wx) : ORIGIN = 0xe000e000, LENGTH = 0x00001000  
  AHB_Peripherals (wx) : ORIGIN = 0x50000000, LENGTH = 0x00200000  
  APB1_Peripherals (wx) : ORIGIN = 0x40080000, LENGTH = 0x00080000  
  APB0_Peripherals (wx) : ORIGIN = 0x40000000, LENGTH = 0x00080000  
  GPIO (wx) : ORIGIN = 0x2009c000, LENGTH = 0x00004000  
  AHBSRAM1 (wx) : ORIGIN = 0x20080000, LENGTH = 0x00004000  
  AHBSRAM0 (wx) : ORIGIN = 0x2007c000, LENGTH = 0x00004000  
  RAM (wx) : ORIGIN = 0x10000000, LENGTH = 0x00008000  
  FLASH (rx) : ORIGIN = 0x00000000, LENGTH = 0x00080000  
}
```

sanbrother, 2020-09-24:
Sorry, I didn't read README carefully.

build_threadx.bat and build_threadx_sample.bat works, I should figure out how to compile with my IDE

closed 2020-09-24 by sanbrother

goldscott, 2020-09-26:
Hi @sanbrother - we used the same (M3) linker script for the M0/M3/M4. We're planning on cleaning this up and making much simpler scripts soon!

sanbrother, 2020-09-26:
@goldscott Thank you so much, I am glad to hear that.