eclipse-threadx/threadx #41: TX_NO_MEMORY is equivalent to TX_START_ERROR in tx_api.h

xiongyu0523, 2020-09-26:
Is this bug?

```
#define TX_SUCCESS                      ((UINT) 0x00)
#define TX_DELETED                      ((UINT) 0x01)
#define TX_NO_MEMORY                    ((UINT) 0x10)
#define TX_POOL_ERROR                   ((UINT) 0x02)
#define TX_PTR_ERROR                    ((UINT) 0x03)
#define TX_WAIT_ERROR                   ((UINT) 0x04)
#define TX_SIZE_ERROR                   ((UINT) 0x05)
#define TX_GROUP_ERROR                  ((UINT) 0x06)
#define TX_NO_EVENTS                    ((UINT) 0x07)
#define TX_OPTION_ERROR                 ((UINT) 0x08)
#define TX_QUEUE_ERROR                  ((UINT) 0x09)
#define TX_QUEUE_EMPTY                  ((UINT) 0x0A)
#define TX_QUEUE_FULL                   ((UINT) 0x0B)
#define TX_SEMAPHORE_ERROR              ((UINT) 0x0C)
#define TX_NO_INSTANCE                  ((UINT) 0x0D)
#define TX_THREAD_ERROR                 ((UINT) 0x0E)
#define TX_PRIORITY_ERROR               ((UINT) 0x0F)
#define TX_START_ERROR                  ((UINT) 0x10)
```

goldscott, 2020-09-26:
@xiongyu0523 - this is not a bug. No function returns both of those symbols (in fact, only tx_timer_create can return TX_START_ERROR).

xiongyu0523, 2020-09-26:
Is there a particular reason we put it in between 0x01 and 0x02 error code there and why not just assign an unique code for this error?

goldscott, 2020-09-26:
@xiongyu0523 - I'm not sure why it's between 0x01 and 0x02, nor why they are not unique. I suspect it is a legacy issue from before ThreadX version 5.0 (which came out in 2005), because these values are the same in v5.0.

xiongyu0523, 2020-09-26:
@goldscott since threadx is still an actively maintained project, can you consider to improve it for a better code quality, when there is no solid reason we need keep it as it is?

goldscott, 2020-09-27:
@xiongyu0523 we need to keep it as is for backwards compatibility.

amgross, 2020-11-26:
> @xiongyu0523 - this is not a bug. No function returns both of those symbols (in fact, only tx_timer_create can return TX_START_ERROR).
I have to say it is really annoying issue, as now you have more places you are returning this status (at threadx modules).
in fact, _txm_module_manager_thread_create can return both errors...

The very basic thing I think is to put at tx_api.h the TX_NO_MEMORY and TX_START_ERROR one next to the other, so when searching for 0x10 there will see also TX_NO_MEMORY.
also if you can just stop using TX_START_ERROR it will be good.

goldscott, 2020-11-30:
@amgross - I would like to change these symbols and make them unique! But we cannot break backwards compatibility.
Modules does add more return codes to some APIs. Modules was added to ThreadX 20 years after it was created, so there were some complications on how to do error handling. 
Where does txm_module_manager_thread_create return TX_NO_MEMORY?

amgross, 2020-12-01:
@goldscott , the minimum I want that you will move:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common/inc/tx_api.h#L154
to be in the order, next to:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common/inc/tx_api.h#L169
so when someone gets error code 0x10 and will look in the header file he will right away recognize there are two such errors. But I would like to just step by step changing all the places you returned TX_START_ERROR with something else.

here is where txm_module_manager_thread_create can return no memory:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_thread_create.c#L330-L333
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_object_allocate.c#L101
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common/src/tx_byte_allocate.c#L263

goldscott, 2021-01-20:
Ah, thank you for pointing that out. 

I have moved TX_NO_MEMORY to be next to TX_START_ERROR so they are in numerical order now. This will be seen in our next release.

closed 2021-01-20 by goldscott