eclipse-threadx/threadx #44: riscv gnu port

saw235, 2020-10-27:
Are there any plans for a risc-v gnu port? I am trying to port the risc-v iar port into gnu but is running into issues mapping the iar directives as well as understanding in general what needs to be ported.

The things that are ported between each platform seems to be very inconsistent (or not well documented) unlike FreeRTOS which has a well defined set of things that needs to be ported in each platform. Particularly, there's no template for tx_port.h and each of the ports seems to port a different set of threadx api.

goldscott, 2020-10-27:
All you really need to do is translate the assembly from IAR to GCC. ThreadX has the same API across all platforms.

Yes, at some point (probably next year) we'll be porting RISC-V to GCC.

saw235, 2020-10-27:
I see. But how do I know what some of these directives do? 

For example: 
PUBLIC seems to easily translate to `.global` here, but EXTWEAK and REQUIRE doesn't seems to map easily to anything.
```
    PUBLIC  _tx_timer_interrupt_handler
    PUBLIC  __minterrupt_000007
    EXTWEAK __require_minterrupt_vector_table
_tx_timer_interrupt_handler:
__minterrupt_000007:
    REQUIRE __require_minterrupt_vector_table
```

There's also some references to external symbols like `__get_interrupt_state` and `__istate_t` which I couldn't wrap my head around what it is actually doing. In some other gnu ports, it is just saving the masked `mstatus`.
 
```
#define TX_INTERRUPT_SAVE_AREA                  __istate_t interrupt_save;
#define TX_DISABLE                              {interrupt_save = __get_interrupt_state();__disable_interrupt();};
#define TX_RESTORE                              {__set_interrupt_state(interrupt_save);};
```

goldscott, 2020-10-27:
Do you really want to do the gnu port yourself?

You'll have to look in the assembler documentation for these directives.

These are intrinsic functions provided by the compiler. 

IAR and GCC will have different directives and intrinsic names, so you'll have to look in their respective documentation for the equivalents.

mrdiogodias, 2021-05-19:
Hey @saw235 
Did you manage to do the port?

mrdiogodias, 2021-06-04:
In case someone comes across this question, I've managed to the[ port to gnu](https://github.com/mrdiogodias/threadx/tree/riscv_gnu). I'm not sure it is 100% correct but it worked well for me and maybe it can help someone out there.

Linjieqiang, 2021-12-13:
Any news? @TiejunMS

TiejunMS, 2021-12-14:
> 

Nope.

goldscott, 2021-12-14:
@hihigupt any comment on this?

Linjieqiang, 2021-12-17:
https://github.com/azure-rtos/threadx/pull/153 @goldscott

goldscott, 2022-10-28:
@Linjieqiang @mrdiogodias @saw235 - hi all, a 32-bit and 64-bit GNU RISC-V port is coming soon. It's in the testing phase now.

closed 2022-10-28 by goldscott