eclipse-threadx/threadx #48: Missing error check of function tx_thread_identify()

Hxinrong, 2020-11-09:
Hi,
As described in the ThreadX documentation, function tx_thread_identify() retrieves pointer to currently executing thread, returns a pointer to the currently executing thread if there is a thread is executing, and returns TX_NULL if not. However, as shown in the following code, a missing error check take places, and then the return value is used.
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/utility/rtos_compatibility_layers/FreeRTOS/tx_freertos.c#L211-L213
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/utility/rtos_compatibility_layers/FreeRTOS/tx_freertos.c#L501-L503
Moreover, in this file, there are also some other call sites of function tx_thread_identify() appearing the same problem.

goldscott, 2020-11-09:
Hi @Hxinrong - thanks for bringing this up. We are looking into it.

goldscott, 2020-11-12:
There is the assumption that the application is calling the APIs from the proper context. It appears that FreeRTOS doesn’t do much if any calling context checking, so we decided not to add any additional checks to the FreeRTOS API layer.

closed 2020-11-12 by goldscott