eclipse-threadx/threadx #49: Missing error check of function tx_thread_resume()

Hxinrong, 2020-11-10:
Hi,
As described in the ThreadX documentation, function tx_thread_resume() resumes or prepares for execution a thread that was previously suspended by a tx_thread_suspend call, returns TX_SUCCESS if there is a successful thread resume, and returns error values if not. However, as shown in the following code, a missing error check take places.
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/common_modules/module_manager/src/txm_module_manager_start.c#L220-L221

Furthermore, the return values of function tx_thread_resume() from other call sites are checked. See the following code, for example.
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/utility/rtos_compatibility_layers/FreeRTOS/tx_freertos.c#L347-L351
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/utility/rtos_compatibility_layers/FreeRTOS/tx_freertos.c#L425-L428

Also, in the same file, maybe the following code about function tx_mutex_put() is needed an error check.
https://github.com/azure-rtos/threadx/blob/6773d468ae877ce36feda1c86842510fb33501a3/common_modules/module_manager/src/txm_module_manager_start.c#L217-L218

goldscott, 2020-11-11:
Thanks for these updates, @Hxinrong. We are still looking into it.

goldscott, 2020-11-12:
We will add error checking to the txm_module_manager_start function. Thanks for your report!

goldscott, 2020-12-10:
Hi @Hxinrong, thank you for your observation regarding error checking. The general principal that we follow is that we perform error checking (when not purposely disabled by TX_DISABLE_ERROR_CHECKING) for situations where the developer is dynamically interacting with ThreadX. For internal calls to the ThreadX API that we know are safe, we do not perform the error checking. Doing so would add additional code and also introduce additional overhead. In the specific case of the ThreadX Module manager identified in the issue we know the call is safe and in fact we are using an internal access function _tx_thread_resume, rather than the publicly exposed API tx_thread_resume, which has no basic usage error checking.

closed 2020-12-10 by goldscott