eclipse-threadx/threadx #51: lose of 4 bytes on the stack when the stack start at aligned address and its size is aligned

amgross, 2020-11-24:
when you are calculating the thread stack end (at thread_create or module_manager_thread_create or module_manager_internal_load), you are calculating it as stack start + stack size -1, which gives you the upper boundary of the stack.

the problem is that in the stack_build (/module_manager_stack_build) porting functions (at both monolithic and modules) you are taking the initial stack pointer as the stack end calculated above, with alignment.
this is bad because the initial stack pointer should be one address **bigger** than the biggest address of the stack.

here is example that will explain this:
let say that we need to be align to 4 bytes, stack_size = 8 and stack_start = 0.
at your calculation: stack end will be at '0+8-1 = 7,' and the initial stack pointer will be at '7 aligned to 4 = 4'.
but if you will put the stack pointer at 8 (which will save 4 bytes), nothing would be put at address 8 because the compiler (at least the arc compiler I am working with, but I think all compilers are working like this) not supposed to put anything at the address that the stack pointer is pointing to when function was called, but is substracting the stack pointer by the number of bytes he need to use (let say 4, so now the stack pointer is 4) and there it is storing the data.

(the change can be done by changing the stack end calculation or [less good] by adding one to the stack pointer before setting its alignment)  

I am using threadx modules 6.1.1, I will be happy if you can fix it in 6.1.x version

goldscott, 2020-11-24:
Hello @amgross, 

The tx_thread_create function is generic C code and is processor/alignment/compiler agnostic. All it is doing is calculating the end address of the stack, and that calculation is correct. 

The tx_thread_stack_build function is where the alignment takes place. Let's say the user passes 15 as the stack size. If stack_start address is 0, stack_end is 14. But this address is not aligned, so tx_thread_stack_build will align it (in this case to a 4 byte boundary), so the stack pointer address will be 12. We cannot go the other way and align it to address 16, because that would violate the stack size.

We choose to take the conservative approach and keep the stack pointer within the stack boundaries (i.e. between stack_start and stack_end). We do not want the stack pointer pointing to address 8 if stack_end is address 7. So depending on the stack size you pass to tx_thread_create, you will lose anywhere from 1 to ALIGNMENT bytes in order to properly align the stack.

amgross, 2020-11-25:
Hello @goldscott 
I understand your point that you prefer the conservative approach and not the RAM saving approach, but still I think it is overkill. from what I see when initializing stack it is common to initialize it to linker variable _estack, which located outside the stack. Also I see in the ARC HW stack checking that STACK_BASE (which you are calling stack_end) is actually outside the stack, so I don't see good reason why not to put also stack_end outside the stack (or at least put the stack pointer there).

JakeSays, 2020-11-27:
@amgross Can you explain to me why this conservative approach is overkill? Personally I would prefer to lose a few bytes over the potential risks associated with such a design change. Keep in mind that this change has the potential to affect every port.

amgross, 2020-11-29:
Of course that if you have a lot of ram it is not an interesting enhancement. but if you will look you will see that the first bytes of the stack are not in use any way it threadx (not counting linux/win ports that I don't sure about them), for example:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports/risc-v32/iar/src/tx_thread_stack_build.s#L228
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports/cortex_a7/gnu/src/tx_thread_stack_build.S#L160
so adding more unused bytes looks to me just too much.

amgross, 2020-11-29:
just now I saw that also in part of your code you assume that stack_end is pointing outside the stack:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_thread_create.c#L159

(as you are nit using <=)

you have there some bugs I think, opening for that different issue

goldscott, 2020-12-02:
We're going to maintain our conservative approach and keep things how they are. Some of these implementations have to do with legacy reasons and maintaining backwards compatibility with older versions of ThreadX. You're worried about 4 bytes of wasted stack space, but you're also using Modules, which uses a lot more memory than regular ThreadX...

amgross, 2020-12-06:
hi @goldscott 
so you can close this issue

closed 2020-12-07 by goldscott