eclipse-threadx/threadx #53: doing in two lines what can do in one

amgross, 2020-11-26:
the second line here is nonsense (comparing the interrupt register with the thread pointer) and suppose to be alwais branch:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports/arc_em/metaware/src/tx_thread_context_restore.s#L165-L166

I think you can replace it with:
`breq    r2, 0, __tx_thread_preempt_restore       ; If not set, preempt executing thread`

goldscott, 2020-11-30:
Hi @amgross,

I think the current code is correct:

brne r2, 0, __tx_thread_no_preempt_restore ; If set, don't preempt executing thread
brne r0, r4, __tx_thread_preempt_restore ; Not equal, preempt executing thread

The first branch checks for the _tx_thread_preempt_flag being set (non-zero). If it is set, we want to always return to the interrupted thread – regardless if there is a newer, higher-priority thread ready as a result of the ISR. The interrupt return to thread code starts at label __tx_thread_no_preempt_restore.

The next compare looks at _tx_thread_current_ptr vs. _tx_thread_execute_ptr. If they are not equal, then we will preempt the currently running thread.

closed 2020-11-30 by goldscott

amgross, 2020-12-01:
oops, you are right, my mistake