eclipse-threadx/threadx #54: make changing thread struct easier

amgross, 2020-11-26:
for now, changing the thread struct (for example by changing TX_THREAD_EXTENSION_1) is very complicated, because in the porting assembly files (where I looked) you put hardcoded the struct element offsets, for example here:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L511

please make it in other way so changing the struct won't enforce changing in all the porting files

JakeSays, 2020-11-26:
@amgross why would it have to be changed in all the porting files? It's unique to a specific port.

amgross, 2020-11-26:
I meant in all the porting files of this processor-compiler. For example, adding something to TX_THREAD_EXTENSION_0 in cortex m3 iar porting, will need to change:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L307
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L315
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L398
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L399
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L402
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/ports_module/cortex-m3/iar/module_manager/src/tx_thread_schedule.s#L403

there are more in this file. I don't find in other files, but maybe there are more in other files that I missed and that is exactly the problem - I don't want to need to search all your porting files where you used offsets that I need to change, and to be worried maybe I missed something.

JakeSays, 2020-11-26:
Well, there's really not much else one could do. Those are the offsets that are required.

Also it's not my port.

goldscott, 2020-11-30:
Hello @amgross -  we have to hard-code offsets into the assembly files because the assembler doesn't know about structs or their layout. You should not use TX_THREAD_EXTENSION_x. Please use TX_THREAD_USER_EXTENSION. This puts everything at the bottom of the thread struct.

closed 2020-11-30 by goldscott