eclipse-threadx/threadx #55: Support the option that _txm_module_kernel_call_dispatcher is part of the module code

amgross, 2020-11-27:
hello
in the non common (non porting) code, yoe are assuming that _txm_module_kernel_call_dispatcher is global variable that holding the pointer to the dispatcher function.
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/inc/txm_module.h#L521-L524

In my system, the dispatcher function will be part of the module code (for example, if I want to do something each time before running the real dispatcher), and hence known to the module at compilation time. of course I can initialize the global at _txm_module_thread_shell_entry with the pointer to my dispatcher function, but I want to not use this global (saving a little run time, saving ram space and less mistakes as it can't get change at run time).

I don't want to touch in my system your common code, so I won't have problems with updating each time to yours last version. can you surround this extern with:
`#ifndef KERNEL_CALL_DISPATCHER`
`#define KERNEL_CALL_DISPATCHER`
`...`
`#endif`
Or put this extern at the porting file (which I don't care to change) or find other solution for me?

goldscott, 2020-11-30:
Hi @amgross - in **txm_module_port.h** you can change the dispatcher function in the `TXM_MODULE_MANAGER_THREAD_SETUP` macro. Does that work for you?

amgross, 2020-12-01:
That works but still can't get rid of this global.

goldscott, 2020-12-01:
All of the module kernel calls use this variable, so it has to stay. We don't foresee people trying to override the dispatcher.

closed 2020-12-01 by goldscott