eclipse-threadx/threadx #56: wrong checking if thread stack overlaps the other

amgross, 2020-11-29:
when you are checking if one stack overlaps the other,
at one point you assume that stack_end is pointing outside the stack:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_thread_create.c#L159

few lines bellow it you assume that the end of the stack is pointing to part of the stack:
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_thread_create.c#L172
and then you are assuming some weird thing (it is bug, need to be <= not matter what you assume):
https://github.com/azure-rtos/threadx/blob/609c75a2e51735fa3ad894e0caef595ab71ba6c9/common_modules/module_manager/src/txm_module_manager_thread_create.c#L176
and you are not checking the option that:
stack_start < next_thread -> tx_thread_stack_start && next_thread -> tx_thread_stack_end < stack_end

goldscott, 2020-11-30:
@amgross - you are right. I have fixed it. How does this look:

```
        /* Check the stack pointer to see if it overlaps with this thread's stack.  */
        
        /* There are four cases the stacks can overlap:
           1. stack_start is less than tx_thread_stack_start; and stack_end is between tx_thread_stack_start and tx_thread_stack_end.
           2. stack_start is between tx_thread_stack_start and tx_thread_stack_end; and stack_end is greater than tx_thread_stack_end.
           3. stack_start and stack_end are both between tx_thread_stack_start and tx_thread_stack_end.
           4. stack_start is less than tx_thread_stack_start and stack_end is greater than tx_thread_stack_end.
           
           We can solve for cases 1-3 by checking if stack_start is between tx_thread_stack_start and tx_thread_stack_end OR stack_end
           is between tx_thread_stack_start and tx_thread_stack_end. 
           We can solve case 4 by checking if tx_thread_stack_start is between stack_start and stack_end.
        */
        
        if ( ( (((UCHAR *) ((VOID *) stack_start)) >= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_start))) &&
               (((UCHAR *) ((VOID *) stack_start)) <= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_end))) ) || 
             ( (((UCHAR *) ((VOID *) stack_end)) >= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_start))) && 
               (((UCHAR *) ((VOID *) stack_end)) <= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_end))) ) )
       {
           /* Either stack_start or stack_end is in the existing thread's stack, clear the stack pointer to 
              force a stack error below.  */
            stack_start =  TX_NULL;
            break;
       }
        
        if( (((UCHAR *) ((VOID *) stack_start)) <= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_start))) && 
            (((UCHAR *) ((VOID *) stack_end)) >= ((UCHAR *) ((VOID *) next_thread -> tx_thread_stack_end))) )
       {
           /* The proposed stack envelopes the existing thread's stack, clear the stack pointer to 
              force a stack error below.  */
            stack_start =  TX_NULL;
            break;
       }
```

amgross, 2020-12-01:
hi
better solution should be:
`if (stack_start<=next_thread -> tx_thread_stack_end && stack_end >= next_thread -> tx_thread_stack_start )`
I **think** it covers all four options

goldscott, 2020-12-01:
Hi @amgross - your solution is better. I've made this fix internally and we will update it on our next release. Thanks!

closed 2020-12-01 by goldscott

amgross, 2021-02-11:
hi @yuxin-azrtos , its a bug, not enhancement.