eclipse-threadx/threadx #58: Exporting the Trace Buffer without development tools

pioneerAlone, 2020-12-07:
I have a question for TraceX.
Because the platform I'm using can't debugging the device with any development tools.
When the program is running, can traceX buffer content be entered into a file periodically for override saving?
If I can do this, I need to pay attention to something.thanks~~

goldscott, 2020-12-07:
TraceX places data into a circular buffer. You can stop the trace and then save the buffer to a file.

closed 2020-12-08 by goldscott

pioneerAlone, 2020-12-09:
Hi, the module I use has an MMU unit, and the kernel and the user layer are memory mapped. When I call tx_trace_enable() in a thread, it returns 0x1D==》TX_NOT_AVAILABLE. What does this mean? 
Can't I use this feature?

goldscott, 2020-12-09:
answering in https://github.com/azure-rtos/threadx/issues/59.