eclipse-threadx/threadx #59: call tx_trace_enable() return 0x1D

pioneerAlone, 2020-12-09:
Hi, the module I use has an MMU unit, and the kernel and the user layer are memory mapped. When I call tx_trace_enable() in a thread, it returns 0x1D==》TX_NOT_AVAILABLE. What does this mean? 
Can't I use this feature?

goldscott, 2020-12-09:
Hi @pioneerAlone - ThreadX does not care if you are using an MMU. You should be able to use trace... Function tx_trace_enable does not return 0x1D (TX_NOT_AVAILABLE). Can you step through the function and tell me where it's returning?

yuxin-azrtos, 2021-01-09:
@pioneerAlone Happy new year! Did you get a chance to investigate the return value 0x1D as goldscott suggested? Please let us know. Thanks!

goldscott, 2021-01-20:
Closing. Please re-open if you have more info to add.

closed 2021-01-20 by goldscott