eclipse-threadx/threadx #60: Caveats when converting tx_initialize_low_level.S from Assembly to C

coder137, 2020-12-13:
I was able to convert the `tx_initialize_low_level.S` threadx startup file from Assembly to C for my personal project [Link here](https://github.com/coder137/STM32-Repo/blob/master/Template/Minimal_Threadx/application/tx_initialize_low_level.c).

After going through the entire process I have a few questions to ask

1. The C version of the code is around 32 bytes smaller than the Assembly version of the same (when compiling with Os). Is there any reason/caveats for using Assembly over C?
2. This is assuming that my above conversion is fine in the long run. Currently, I have only tested threadx tasks and that seems to work fine. Would it be possible to review [this file](https://github.com/coder137/STM32-Repo/blob/master/Template/Minimal_Threadx/application/tx_initialize_low_level.c) and provide any comments here if there are some considerations to be made?
3. I noticed that there are certain handler functions that are being declared but I cannot find their usage anywhere. Why have they been defined?
- `__tx_IntHandler`
- `__tx_SysTickHandler` however `SysTick_Handler` (which runs the same chunk of code) is used since it overrides the weak linkage of the startup function.
- `__tx_BadHandler`
- `__tx_HardfaultHandler`
- `__tx_SVCallHandler`
- `__tx_NMIHandler`
- `__tx_DBGHandler`

Currently, I haven't tried to make too many changes to the C version of the `tx_initialize_low_level` and have translated it from assembly to C as closely as possible.
Any pointers or clarification to the above questions would be great!

goldscott, 2020-12-22:
1. The compiler likely optimized some of the register loading and combined the simple while(1) handlers. What .S file are you using? We used assembly because this comes from our legacy ports where this file has always been written in assembly. You are certainly free to use C!
2. This function is only called once when ThreadX is initializing, so if the unused memory pointer, systick, and priorities are set up correctly, you should be good to go.
3. I'm not sure why we defined these tx-prefixed handlers either... they aren't used. I think they ought to be removed. In other cortex-m ports, we use the SysTick_Handler rather than renaming it (e.g. https://github.com/azure-rtos/threadx/blob/master/ports/cortex_m33/gnu/src/tx_initialize_low_level.S#L173-L180). 

I think you're on the right track. You are free to define those unused handlers however you want based on the needs of your application.

coder137, 2020-12-24:
As per your question I am using [this file](https://github.com/coder137/STM32-Repo/blob/master/Template/Minimal_Threadx/application/tx_initialize_low_level.S) adapted from the sample demo examples for Cortex M4.

Thank you for your detailed answer! These clarifications help a lot.

goldscott, 2021-01-20:
Closing, but feel free to reopen if you have more questions.

closed 2021-01-20 by goldscott