eclipse-threadx/threadx #62: Overwrite all the unused any more system stack

amgross, 2020-12-15:
I suggest to add in the beginning of the scheduler the line:
`mov sp, _estack`
as the scheduler is never returning there is no point to save on it all the call stack of the initialization of the processor and the initialization of the kernel.
that is important because the interrupts are using this stack.
https://github.com/azure-rtos/threadx/blob/848e7d81699393967f25c4a4bf980667a54001d2/ports/arc_em/metaware/src/tx_thread_schedule.s#L84-L88

goldscott, 2020-12-17:
Hello @amgross - we will fix this next time we update the ARC ports. I'll keep this open and let you know when we make the fix. Thanks!

amgross, 2021-04-04:
Hi @goldscott ,
you updated ARC ports and I don't see it was done.
Am I mistaken?

andrejm-rtos, 2021-09-13:
This has been implemented and it will be part of the next release.

andrejm-rtos, 2021-10-04:
This has been fixed and it will be included in the October 2021 Azure RTOS release.

closed 2021-10-04 by andrejm-rtos