eclipse-threadx/threadx #63: make object checking a bit more faster

amgross, 2020-12-21:
I suggest to check if the pointers are NULL just after checking if they are in the region.
https://github.com/azure-rtos/threadx/blob/b0e9b132b59ab703f19b4588ed785c9c8e7a1072/common_modules/module_manager/inc/txm_module_manager_util.h#L85-L105
 that won't cause problems as the pointers are not in use here, and it will be faster because in most of the time they are inside the regions (and in c, conditions are checked from left to right).

goldscott, 2020-12-22:
Hi @amgross - thank you for the suggestion! I didn't know about the guaranteed left to right checking; I learned something new. I'll add this change to my to-do list.

MaJerle, 2020-12-22:
This can help you.

https://stackoverflow.com/questions/2456086/order-of-execution-for-an-if-with-multiple-conditionals

goldscott, 2021-03-19:
This improvement has been added internally and will be available in our next release.

closed 2021-03-19 by goldscott