eclipse-threadx/threadx #64: Failed to build

simon13579, 2020-12-28:
Hello,

I follow the README to build the sample, but it report the error when I execute the command "cmake --build ./build", could you please help me check? Thanks!

Thanks,
Simon
[image]

TiejunMS, 2020-12-29:
I guess the cmake version is less than 3.13 so `-B` option does not work. You can use these commands instead. In `threadx` folder, run
``` shell
mkdir -p build
cd build
cmake -DCMAKE_TOOLCHAIN_FILE=cmake/cortex_m4.cmake -GNinja ..
cmake --build .
```

@ryanwinter , do you think we should require a minimal cmake version of 3.13 or update the readme instructions?

goldscott, 2020-12-31:
Hi @simon13579 - have you been able to build after using @TiejunMS suggestion?
Either update your cmake or use the commands above. 

Thanks!

simon13579, 2021-01-01:
Hello @goldscott @TiejunMS ,

I am able to bulid after following @TiejunMS  suggestion. Thanks!

Best Regards,
Simon

closed 2021-01-01 by simon13579