eclipse-threadx/threadx #65: don't allocate space to filex when not using it

amgross, 2020-12-29:
can you add some flag such as TXM_MODULE_ENABLE_FILEX that you have in modules, and add this pointer to the thread struct just when it defined?
https://github.com/azure-rtos/threadx/blob/b0e9b132b59ab703f19b4588ed785c9c8e7a1072/common/inc/tx_api.h#L422
I suppose that most of threadx user don't use filex

goldscott, 2020-12-31:
Hi @amgross - it is probably easier to waste a pointer in each thread struct rather than having the user #define some symbol to enable FileX and rebuild the ThreadX library. I'll discuss with the team and get back to you.

yuxin-azrtos, 2021-02-05:
@amgross As this request is now tracked internally, I am closing the issue. Feel free to reopen it if you have additional thoughts on this. Thanks!

closed 2021-02-05 by yuxin-azrtos

amgross, 2021-03-07:
thanks for implementing :)