eclipse-threadx/threadx #67: cant remove unused code at threadx modules by the linker

amgross, 2020-12-30:
in the monolitic threadx, 'smart' linkers can remove code sections that are not in use, for example if I am not usinf some info_get function.

in threadx modules that can't do such things because all functions are referenced from the kernel dispatcher.

can you surround each case in the dispatche with somthing like:

`#ifndef TXM_BLOCK_POOL_CREATE_CALL_NOT_IN_USE`
`  case TXM_BLOCK_POOL_CREATE_CALL:`
`  {`
`    ...`
`  }`
` #endif`
 
and then the user can remove if he wants the references to the function he not using.
I think it is important feature for picokernel to remove unused code

goldscott, 2021-01-08:
Hi @amgross - internally, we're discussing possible solutions. I'll keep this ticket open and we'll keep you posted.

amgross, 2021-02-14:
Hi @goldscott , any updates on this ticket?

amgross, 2021-02-17:
hi @goldscott 
this feature is important also in security view to give option to block modules from calling unwanted system calls (for example if we want that the module manager will have option to call some XXX_PERFORMANCE_SYSTEM_INFO_GET_CALL system calls and not the modules)

goldscott, 2021-02-17:
We'll be adding compile time options to the dispatcher to allow you to choose what functions modules can call.

amgross, 2021-02-18:
thanks, it will be very helpful

goldscott, 2021-03-24:
Hi @amgross - we have added this feature and it will be available in the next release.

amgross, 2021-04-04:
Hi,
I see you done it in 6.1.6?
So I am closing this.

closed 2021-04-04 by amgross