eclipse-threadx/threadx #68: optional race condition

amgross, 2021-01-03:
I think that if before the clri we will get interrupt that changing the thread to execute, this thread won't execute if nothing expired. (because you are doing here context save/restore just if something expired).
https://github.com/azure-rtos/threadx/blob/b0e9b132b59ab703f19b4588ed785c9c8e7a1072/ports/arc_em/metaware/src/tx_timer_interrupt.s#L81-L90

because the timer interrupt was set to the lowest priority that can happen.

it is identical to the race condition you are asking to prevent here:
https://github.com/azure-rtos/threadx/blob/b0e9b132b59ab703f19b4588ed785c9c8e7a1072/ports/arc_em/metaware/readme_threadx.txt#L166-L167

goldscott, 2021-01-12:
Hi @amgross - Looks like we need to fix the tx_timer_interrupt.s file. There is a very tight window between the vector fetch and execution of the clri instruction. To fix it, perhaps the easiest option is to look at __tx_thread_current_ptr and _tx_thread_execute_ptr at line 178 (in the 6.1.3 file). If they are not equal, we jump to __tx_something_expired label.

amgross, 2021-01-13:
Hi @goldscott  - doesn't it cleaner to put 1 in the expiration flag when they are not equal? (it is same code size, and very little more run time in this race condition).
that way if someone will add something afterword it won't skip. 

Can you please give me the exact code you are planning to add for solving it?

amgross, 2021-01-25:
Hi @goldscott  , can you give me the exact code for fixing it or estimation when you will fix it?

goldscott, 2021-01-29:
Hi @amgross - we are waiting for tools from Synopsys so we can test these changes. Probably will be available in the March release of ThreadX.

amgross, 2021-04-04:
Hi @goldscott 
I understand you still don't have the tools?
Any time estimations?