eclipse-threadx/threadx #69: Why custom application ISR can't be in same priority as ThreadX-managed ISR?

amgross, 2021-01-11:
Hi,
in ARC-EM port you are claiming:
https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/ports/arc_em/metaware/readme_threadx.txt#L166-L167
I understand why ISRs that not using context save/restore can't be in lower priority - because if such interrupt will be interrupted by higher priority ISR that uses context save/restore that will cause that other thread should execute, the context restore won't return to the scheduler because it is nested interrupt and the lower interrupt won't return to the scheduler because it didn't using context restore.

but I don't understand why they can't be in the same priority - then nested interrupts won't occur and the ThreadX-managed ISR will return to scheduler if needed (and indeed maybe will interrupt right after that by the other interrupt)

goldscott, 2021-01-12:
Hi @amgross - provided there is no nesting possible at the same priority level, then yes, it would be possible to have unmanaged ISRs equal in priority to the highest priority ThreadX ISR.

amgross, 2021-01-13:
Hi @goldscott  - I think there is no way. according [synopsys](https://arc-labs.readthedocs.io/en/latest/labs/level1/lab4.html):

> Interrupt handlers can be preempted by higher-priority interrupts

can you confirm to me it is really the case and it will work OK?

andrejm-rtos, 2021-10-04:
This is the case for ThreadX. This is also how we understand the ARC architecture. 

But for precise details on the ARC architecture, please check with Synopsis.