eclipse-threadx/threadx #70: Documentation: issue with FPU enabled thread stack frame in readme_threadx.txt

newbrain, 2021-01-13:
In all the readme_threadx.txt for cortex-m with an FPU the interrupted stack frame with FPU enabled looks like this:
https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/ports/cortex_m33/gnu/readme_threadx.txt#L66-L115

Looking at the PendSV_Handler implementation in tx_thread_schedule.S for e.g. the M33 port (same code as for M7 and M4):
https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/ports/cortex_m33/gnu/src/tx_thread_schedule.S#L143-L150

The correct stack order should be as follows, from lower offsets to higher ones:

- LR: Interrupted LR (LR at time of PENDSV)
- S16-31: SW stacked FPU registers
- R4-11: SW stacked GP registers
- R0-3,12,LR,PC,xPSR: HW stacked GP and processor state registers
- S0-15: HW stacked FPU registers
- FPSCR: HW stacked FPU state register

goldscott, 2021-01-15:
Hi @newbrain - you're absolutely correct - we'll update the readmes with the correct stack layout. Thanks for bringing this up!
I think the readmes are from before we started using the hardware lazy stacking.

newbrain, 2021-01-15:
Thanks!
I needed to make sure, since I'm developing a [pyocd](https://github.com/pyocd/pyOCD) plug-in for Azure RTOS ThreadX and need to unwind the stack frame.

goldscott, 2021-03-05:
Fixed in https://github.com/azure-rtos/threadx/releases/tag/v6.1.5_rel

closed 2021-03-05 by goldscott