eclipse-threadx/threadx #72: Defines TX_MISRA_ENABLE and  TX_ENABLE_STACK_CHECKING are not compatible

csrichter, 2021-02-01:
If TX_MISRA_ENABLE and  TX_ENABLE_STACK_CHECKING are both enabled the build fails

tested with m0 port on EWARM toolchain

This is because in tx_api.h the symbols TX_POINTER_TO_ALIGN_TYPE_CONVERT und TX_ALIGN_TYPE_TO_POINTER_CONVERT are not defined if TX_MISRA_ENABLE is defined

Is that expected behaviour? Is stack checking simply not MISRA compatble?

goldscott, 2021-02-03:
Hi @csrichter - right now it looks like these defines are mutually exclusive, which should not be the case. I'll file this as a bug and we'll work on it. Thanks for bringing this up! I'll leave this ticket open until we get a fix released.

yuxin-azrtos, 2021-05-19:
We fixed the code so TX_MISRA_ENABLE and TX_ENABLE_STACK_CHECKING can be used together.  It will be available in the upcoming release.

closed 2021-05-19 by yuxin-azrtos

cffranck, 2026-08-18:
> Is stack checking simply not MISRA compatble?

MISRA rules disallow or discourage many of the operations necessary for stack checking, such as converting between pointer types and integral types. These operations are, at best, implementation-defined behavior, and at worst undefined behavior.

Some of the functionality of the stack checker also implies that undefined behavior has already occurred, making reliable operation of such software checks a matter of wishful thinking.