eclipse-threadx/threadx #73: Define TX_MISRA_ENABLE doesn't work

csrichter, 2021-02-01:
Setting the Define TX_MISRA_ENABLE leads to compile error, because function _tx_misra_ipsr_get() is missing,

Using IAR EWARM toolchain and the M0 Port of threadx with a stm32f091 as target MCU

goldscott, 2021-02-03:
I'm not sure why the tx_misra.s file is not part of the M0 IAR port... Anyway, use the tx_misra.s file in the M3 IAR port.

csrichter, 2021-02-03:
Thanks for the quick answer.
Your suggestion doesn't work though, it leads to multiple compile errors:
[image]

It seems like the m3 assembly uses features not available in the m0

But extracting the function _tx_misra_ipsr_get from tx_misra.s seems to work on the m0 (in combination with the provided tx_misra.c file)

yuxin-azrtos, 2021-05-19:
@csrichter  We do not have MISRA support on Cortex-M/IAR.  If you need that feature, please email azure-rtos-support@microsoft.com.

closed 2021-05-19 by yuxin-azrtos