eclipse-threadx/threadx #74: Custom events configurtion with TraceX

fzawadiak, 2021-02-03:
Don't know where else to put it. TraceX is downloadable now only from the Store. Documentation about defining custom events is wrong, there is no "installation directory" with eltrxcustom.trxc etc. Kind of severly limiting usability of ThreadX. Preferably file would be read from project directory, not installation directory.

yuxin-azrtos, 2021-02-03:
@fzawadiak thank you for the feedback.  We will review it and get it back to you.

goldscott, 2021-02-05:
Hi @fzawadiak - in our transition to using the Store, we didn't account for the lack of access to the install directory when trying to build custom events. I believe we are changing this to allow you to select a folder to use for custom events. Stay tuned for TraceX updates.

jdeere5220, 2021-02-09:
@fzawadiak we have published an update to the Microsoft Store that should correct the problem. Your TraceX should automatically update to version 6.1.4.0, usually within 24 hours. Once your app has been updated, when you first run TraceX you should be prompted to choose a local folder for your example trace data files and your custom event specification file.

goldscott, 2021-02-12:
Thanks @jdeere5220 
I'll close this ticket now.

closed 2021-02-12 by goldscott