eclipse-threadx/threadx #75: prevent user/memory protected module from starving the whole system

amgross, 2021-02-08:
Hi
From what I see, in threadx module each thread can create other thread with highest priority and/or n time slice, which can starve the whole system. Am I missing something?

can I somehow limit the range of treads priorities/time slice in module, or better - define that in module thread can't create other thread with higher priorities?

goldscott, 2021-02-08:
Hi @amgross 

There is an API https://github.com/azure-rtos/threadx/blob/master/common_modules/module_manager/src/txm_module_manager_maximum_module_priority_set.c to limit the maximum priority of a thread that a module can create. 

There is no API for limiting the time slice.

amgross, 2021-02-09:
thank you, I missed it.

isn't it reasonable to limit also the time slice?

goldscott, 2021-02-09:
The issue with limiting time slice is that threads can be created with or without time slices and time slicing only works for threads that have equal priorities.

amgross, 2021-02-12:
By limiting time slice I mean also not allowing non time slice.

for example, maybe I will want that modules priority will be 10<=  and that time slices will be 1<=x<=7, then each module can know that if it has thread in priority 10 so no other module can starve it.

goldscott, 2021-02-12:
We already have the ability to limit the priority of threads within a module, which provides system starvation protection (time domain protection). As for time-slicing, that only applies to threads of the same priority. We don't want to get into the business of deciding for the application how long any thread of the same priority can run.

closed 2021-02-12 by goldscott