eclipse-threadx/threadx #76: Execution profile kit removal

jltrahan-2020, 2021-02-09:
Hi,

We've recently migrated one of our projects from ThreadX v5.8 to Azure RTOS v6.1.  In the process, we've discovered that we had the Execution Profile kit installed.  This was probably forgotten in there from years ago when we first started using ThreadX on the RZ A1 MCU.  Since this is now stable code in production (and execution profiling adds code/delays to every thread/interrupt change), we've decided to remove the profiling code, basically reversing the instructions provided in the comments of tx_execution_profile.c.

However, removing the Execution Profile kit code seems to cause the ARM A9 core (which is what is in the RZ A1) to eventually generate a spurious int_id = 1023.  When IntC_Handler() gets called, this int_id is illegal, and the function traps into a while(1).  I'm unable to understand why this happens.  This occurs whether in v5.8 or v6.1 of ThreadX/Azurt RTOS.  I suspect that the Execution Profile kit masks some issues with interrupt executions, but I'm no expert on ARM architectures.

I've contacted Renesas, but they suggested that I get in touch with the tx maintainers.

Jean-Luc

goldscott, 2021-02-09:
Removing the EPK should no affect - it doesn't do anything with interrupts. You must've set up some hardware timer for the timing, so you can turn that off if it's no longer in use. Was that set up to trigger an interrupt?

jltrahan-2020, 2021-02-10:
Hi Scott,

You raise a very good point, but it's not the issue.  You are correct, I had forgotten to stop the associated H/W timer for this, but it's just a free running timer, which doesn't generate interrupts.  I removed the short bit of code that starts this timer, but the behavior remains the same, as expected (as there are no interrupts for this timer.)

Jean-Luc

jltrahan-2020, 2021-02-10:
You raise a very good point, but it’s not the issue.  You are correct, I had forgotten to stop the associated H/W timer for this, but it’s just a free running timer, which doesn’t generate interrupts.  I removed the short bit of code that starts this timer, but the behavior remains the same, as expected (as there are no interrupts for this timer.)

goldscott, 2021-02-10:
Did you undefine TX_ENABLE_EXECUTION_CHANGE_NOTIFY and rebuild ThreadX?
Can you figure out what/where the interrupt is being triggered?

jltrahan-2020, 2021-02-10:
Yes, I basically reversed the instructions provided in the comments of tx_execution_profile.c, and that includes removing the TX_ENABLE_EXECUTION_CHANGE_NOTIFY define, and I did a rebuild all of the project too.

I'm also currently testing a fix proposed by Renesas.  Apparently there's a known issue with this particular ARM GIC, which causes spurious int_id = 1023.  There's workaround code that I'm trying, but I get another problem with this code: Data Aborts!!  I'll try to get more details shortly.

goldscott, 2021-02-12:
I'm assuming the issue has something to do with the ARM GIC, because removing the EPK shouldn't have any effect on interrupts.

jltrahan-2020, 2021-02-12:
Your assumption may possibly be correct, but it does beg the question: why does it _prevent_ the issue from happening?  My code **with** the Execution Profile kit is perfectly happy, but as soon as I remove it, I get these spurious interrupts.  So, I'm not 100% convinced the Execution Profile does not affect the interrupts.

goldscott, 2021-02-12:
Can you figure out where the interrupt is being triggered? 

Nothing in ThreadX sets up int id 1023... The only interrupt ThreadX needs is a timer interrupt. This is configured in the hwsetup function and uses id 134.

jltrahan-2020, 2021-02-17:
Nothing can actually setup int_id = 1023, as it's an illegal interrupt on this architecture.  The interrupts on this implementation of the A9 core can have values up to 586, this includes all sources, software/hardware/peripheral/etc.

I'm almost convinced that this is due to the ARM errata we talked about earlier, but then Renesas tells me they are pushing this to you guys via their "back channels".  So, I'm kind of stuck between the two of you.

Again, what intrigues me the most, is why the Execution Profile code would prevent this problem?

goldscott, 2021-02-27:
Hi @jltrahan-2020 any updates for us?

jltrahan-2020, 2021-03-01:
I have not had time to look into this in much detail.  I was expecting either Renesas or Microsoft to help me out on this as I am not an expert on ARM.

goldscott, 2021-03-19:
Working on this via email. Closing.

closed 2021-03-19 by goldscott