eclipse-threadx/threadx #78: Why the timer thread can't stop module?

amgross, 2021-02-11:
hi

from what I see, the timer thread can't stop module:

https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/common_modules/module_manager/src/txm_module_manager_stop.c#L156-L161

What is the reason for that?

goldscott, 2021-02-11:
Hi @amgross - I'm not sure why the timer thread can't stop a module. Seems like it would be a useful feature (e.g. be able to start a timer, run a module, and stop the module when the timer expires). I'll try to find out why we don't allow this now.

goldscott, 2021-02-12:
Hi @amgross we don't allow timers to stop modules because the stop function could take a while if a module uses a lot of objects (we delete these objects in the stop function). For performance reasons, we do not want a lot of processing or blocking in the timer thread, so it's best if the stop function is called elsewhere.

closed 2021-02-12 by goldscott

amgross, 2021-02-12:
Hi @goldscott , it looks to me an important feature, can't you add some compilation flag that will allow the programmers to decide if they want to allow it? (even if will limit some things)