eclipse-threadx/threadx #80: cleaning block/byte pool region upon release

amgross, 2021-02-24:
Hi
As far as I see there is no built in option to clean the region of block/byte pool upon release.

I think it is very important compile option. especially when using memory protection so it is data leak source (also, in threadx module the modules manager is releasing the pools including the thread struct, so it is problematic to clean it from the module).

I don't think it is a lot of effort to add this option. I understand it will take a runtime when releasing so I suggest to add it just if some compilation flag is on.

goldscott, 2021-02-27:
Hi @amgross - we're discussing internally. I think this is a good idea.

amgross, 2021-04-04:
Hi @goldscott,
What is the status of this issue?

goldscott, 2021-04-06:
Hi @amgross - some higher priority bug fixes were ahead of this feature request, so this did not make it into the 6.1.6 release.

goldscott, 2021-09-29:
For now, we've updated the ThreadX manual with notes in the `tx_byte_release` and `tx_block_release` APIs suggesting that users may want to clear this memory to prevent data leaks.

closed 2023-03-14 by TiejunMS