eclipse-threadx/threadx #82: Integer underflow upon module stop

amgross, 2021-03-01:
When stopping module, the object list count underflows upon finishing to release the objects (its checking the 0, and then subtracting it):
https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/common_modules/module_manager/src/txm_module_manager_stop.c#L534
hence, if starting the module without unloading it, cause problems (probably exceptions) when it allocates objects:
https://github.com/azure-rtos/threadx/blob/32e3b3b25f4b17ff51ce8a6bad8d9eb32729a0f6/common_modules/module_manager/src/txm_module_manager_object_allocate.c#L110-L130

goldscott, 2021-03-01:
Thank you for finding this, @amgross. I'll close this ticket when the fix is available on github.

goldscott, 2021-03-05:
Fixed in https://github.com/azure-rtos/threadx/releases/tag/v6.1.5_rel

closed 2021-03-05 by goldscott