eclipse-threadx/threadx #83: ThreadX 6.0 errata?

robert-lee2, 2021-03-02:
We are using Azure RTOS ThreadX 6.0 for Cortex M4.  I am looking for an errata / list of defects found / fixed since 6.0 was released.  I want to evaluate the defects to see if we need to upgrade before we release.  Where can I find an errata?

goldscott, 2021-03-02:
Hi @robert-lee2, see the release notes here for each release to determine defects found and fixed since 6.0 was released: https://github.com/azure-rtos/threadx/releases

This is the closest thing we have to a list of errata.

robert-lee2, 2021-03-03:
We have 6.02.  I can see from the 6.1 release notes that there were "multiple bug fixes", but I cannot find a list of the bugs fixed to determine if we should upgrade because a file we're using is affected.  Am I missing something?

goldscott, 2021-03-03:
Perhaps "multiple bug fixes" is misleading. There really were no bug fixes in 6.0.2. Rather, there were multiple changes made and mentioned right below that, e.g.

Azure RTOS ThreadX 6.0.2 release contains the following changes:

    ports/cortex_m0 - all toolchains: modified comments and whitespace.
        ports/cortex_m0/ac6/example_build/sample_threadx/tx_initialize_low_level.S - remove DWT code.
        ports/cortex_m0/gnu/example_build/tx_initialize_low_level.S - remove DWT code.
        ports/cortex_m0/iar/example_build/tx_initialize_low_level.s - remove DWT code.
    ports/cortex_m3 - all toolchains: modified comments and whitespace.
    ports/cortex_m33 - all toolchains: modified comments and whitespace.
        ports/cortex_m33/ac6/src/tx_thread_context_restore.S - remove execution profile kit call.
        ports/cortex_m33/ac6/src/tx_thread_context_save.S - remove execution profile kit call.
        ports/cortex_m33/ac6/src/tx_timer_interrupt.S - add DSB instruction before returning.
        ports/cortex_m33/iar/src/tx_thread_context_restore.s - remove execution profile kit call.
        ports/cortex_m33/iar/src/tx_thread_context_save.s - remove execution profile kit call.
    ports/cortex_m4 - all toolchains: modified comments and whitespace.
        ports/cortex_m4/ac5/src/tx_thread_context_save.s - fix register names.
    ports/cortex_m7 - all toolchains: modified comments and whitespace.
    common/inc/tx_api.h update product constants.
    common_smp/inc/tx_api.h update product constants.

We will be more clear in future release notes about bugs/changes/additions.

closed 2021-03-03 by robert-lee2