eclipse-threadx/threadx #84: Tickless mode in ThreadX

jmmb13, 2021-03-04:
I'm currently using the ThreadX in a Cortex-M0+, and my application needs to take advantage of the microcontroller's low power modes. So, I extended the timer.h module with the function _tx_timer_tickless_mode. This function is called in the scheduler when there isn't a thread ready to run and replaces the WFI assembly instruction for the low power mode of the MCU.

The _tx_timer_tickless_mode starts searching in the timers entry wrap the minimum remaining ticks (right now, because the insert is always done in the end, the search needs to be in all nodes of the lists, but can be optimized by inserting the timers orderly).

It checks if the ticks are greater than a threshold  (TX_TIMER_ENTRIES, can be changed) after getting the remaining ticks. If it is greater, it decrements the sleep ticks by 1 to wake up one tick before the timer's expiration with the lowest remaining ticks.

In order to do it generic for various boards, I implemented three __weak functions:
 - configPreSleepMode(sleep_time) -> receives the sleep_time in ticks and is responsible for disabling the tick and enable an interrupt to wake up the MCU in time;
 - sleepMode(void) -> It is responsible for using the low power modes of the MCU;
 - configPosSleepMode(void) -> Enables the tick, and does the configuration after sleep if needed, returns time passed in ticks;

Now, the remaining ticks are refreshed in the timers entry wrap with the ticks passed and then uses an auxiliary array similar to timers entry wrap and is done the reinitialization of all the timers and is copied in the end this list to the actually timers entry wrap.

**tx_thread_schedule.s**

```asm
__tx_ts_wait
    CPSID   i                                       ; Disable interrupts
    LDR     r1, [r2]                                ; Pickup the next thread to execute pointer
    STR     r1, [r0]                                ; Store it in the current pointer
    CMP     r1, #0                                  ; If non-NULL, a new thread is ready!
    BNE     __tx_ts_ready                           ;   
    IF :DEF:TX_ENABLE_WFI
    PUSH    {r0-r3}                                 ; Save of r0 to r3 registers
    BL _tx_timer_tickless_mode                      ; Jumps to the function of tickless mode
    POP    {r0-r3}                                  ; Pop of r0 to r3 registers
    ENDIF
__tx_ts_ISB
    CPSIE   i                                       ; Enable interrupts
    B       __tx_ts_wait                            ; Loop to continue waiting
```

**tx_timer_tickless_mode.c**

```c
/** @brief  Receives the sleep time and returns the actual time
  *
  * Receives the sleep_time in ticks to set in order to MCU wake up and returns the initial time,
  * which will be used in the future to know how many has passed. The tick needs to be disable in 
  * case of the MCU goes into sleep.
  *
  * @param sleep_time_ul is the sleep time in ticks needed to be set to mcu wake up, if sleep time
  * time is ~0 it doesn't go into sleep mode
  *
  * @retval The actual time before sleep in ticks or (ULONG)~0 to don't go to sleep
  */
__weak ULONG configPreSleepMode(ULONG sleep_time_ul){

    if(sleep_time_ul <= (TX_TIMER_ENTRIES<<1) ){
        return (ULONG)(~0);
    }
    return 0;
}

/** @brief  Enters into sleep mode
  *
  * @retval None
  */
__weak VOID sleepMode(VOID){
    __asm volatile ( "WFI" ); /*Wait for interrupt*/
    __asm volatile ( "ISB" ); /*Ensure pipeline is flushed*/
    return;
}

/** @brief  Coniguration pos sleep mode, returns the time passed after sleep in ticks
  * 
  * The tick needs to be enable.
  *
  * @retval returns the time passed after sleep in ticks
  */
[... 226 lines cut ...]
```c
TIMER_DECLARE TX_TIMER_INTERNAL **_tx_timer_list_tickless_start;
TIMER_DECLARE TX_TIMER_INTERNAL **_tx_timer_list_tickless_end;

TIMER_DECLARE TX_TIMER_INTERNAL *_tx_timer_list_tickless_recovery[TX_TIMER_ENTRIES];
````

And to **tx_initialize.c**

```c
    /* Initialize all of the list pointers.  */
    _tx_timer_list_tickless_start =   &_tx_timer_list_tickless_recovery[0];

    /* Set the timer list end pointer to one past the actual timer list.  This is done
       to make the timer interrupt handling in assembly language a little easier.  */
    _tx_timer_list_tickless_end =     &_tx_timer_list_tickless_recovery[TX_TIMER_ENTRIES-((ULONG) 1)];
    _tx_timer_list_tickless_end =     TX_TIMER_POINTER_ADD(_tx_timer_list_tickless_end, ((ULONG) 1));

```
The array _tx_timer_list_tickless_recovery and the start and end pointers can be created locally on the system stack instead of being declared in the tx_timer module.

What's your opinion on this subject?

Thanks for the attention,
Marcelo

goldscott, 2021-03-04:
Hello @jmmb13 - I'm sorry you've done all this work, as later today or tomorrow we are releasing ThreadX 6.1.5 which has some additional low-power utility files which basically do ThreadX timer management, so this will take care of the ThreadX timers and you simply need to add your code to turn peripherals off, optionally implement a hardware sleep timer, and go into sleep mode. Please contact me directly and I'll send you the low power utilities now.

goldscott, 2021-03-05:
Added in https://github.com/azure-rtos/threadx/releases/tag/v6.1.5_rel

closed 2021-03-05 by goldscott