eclipse-threadx/threadx #87: Error compiling for Cortex-M0 with GCC

josesimoes, 2021-03-11:
I'm trying to compile for a Cortex-M0 (actually a Cortex-M0+) and I'm getting these errors:

```
[build] [19/246   4% :: 0.279] Building C object targets/AzureRTOS/RaspberryPi/RPI_PICO/AzureSource/CMakeFiles/threadx.dir/common/src/tx_thread_delete.c.obj
[build] FAILED: targets/AzureRTOS/RaspberryPi/RPI_PICO/AzureSource/CMakeFiles/threadx.dir/common/src/tx_thread_delete.c.obj 
[build] E:\nftools\GNU_Tools_ARM_Embedded\9-2020-q2-update\bin\arm-none-eabi-gcc.exe -DTX_INCLUDE_USER_DEFINE_FILE -I_deps/azure_rtos-src/ports/cortex_m0/gnu/inc -I_deps/azure_rtos-src/common/inc -Itargets/AzureRTOS/RaspberryPi/RPI_PICO/AzureSource/custom_inc -g   -DDEBUG -std=c11 -MD -MT targets/AzureRTOS/RaspberryPi/RPI_PICO/AzureSource/CMakeFiles/threadx.dir/common/src/tx_thread_delete.c.obj -MF targets\AzureRTOS\RaspberryPi\RPI_PICO\AzureSource\CMakeFiles\threadx.dir\common\src\tx_thread_delete.c.obj.d -o targets/AzureRTOS/RaspberryPi/RPI_PICO/AzureSource/CMakeFiles/threadx.dir/common/src/tx_thread_delete.c.obj   -c _deps/azure_rtos-src/common/src/tx_thread_delete.c
[build] C:\Users\jose\AppData\Local\Temp\cccGFaRj.s: Assembler messages:
[build] C:\Users\jose\AppData\Local\Temp\cccGFaRj.s:47: Error: selected processor does not support requested special purpose register -- `mrs r3,PRIMASK'
[build] C:\Users\jose\AppData\Local\Temp\cccGFaRj.s:55: Error: selected processor does not support `cpsid i' in ARM mode
[build] C:\Users\jose\AppData\Local\Temp\cccGFaRj.s:85: Error: selected processor does not support requested special purpose register -- `msr PRIMASK,r3'
[build] C:\Users\jose\AppData\Local\Temp\cccGFaRj.s:159: Error: selected processor does not support requested special purpose register -- `msr PRIMASK,r3'
```

Similiar ones show up with other files.

This goes away it I add `#define TX_DISABLE_INLINE` to tx_user.h.

Seems that something is wrong with the thumb / arm modes...

goldscott, 2021-03-11:
Seems like you don't have the right build settings - you're trying to build in ARM mode rather than THUMB mode.

josesimoes, 2021-03-11:
Here's what I'm using :

`set(CMAKE_ASM_FLAGS " -mcpu=cortex-m0plus -mtune=cortex-m0plus -mthumb " CACHE INTERNAL "asm compiler flags")`

and 

`set(CMAKE_EXE_LINKER_FLAGS " -Wl,--gc-sections -Wl,--no-wchar-size-warning -Wl,--print-memory-usage -mthumb -mcpu=cortex-m0plus -nostartfiles " CACHE INTERNAL "executable linker flags")`

goldscott, 2021-03-11:
Somehow those settings don't seem to be active when actually building the files. 

Have you built the example that is included in threadx\ports\cortex_m0\gnu\example_build?

josesimoes, 2021-03-11:
> Somehow those settings don't seem to be active when actually building the files.

🤔 could be... let me double check this...

josesimoes, 2021-03-12:
@goldscott you are correct! At some point the CMAKE definitions for the compiler where being overriten. 
This is not an issue anymore.

Apologies for taking your time and completly missing that the compiler options where not there at all...

closed 2021-03-12 by josesimoes