eclipse-threadx/threadx #88: tx_initialize_low_level.s:33: undefined reference to `systick_cycles' collect2.exe: error: ld returned 1 exit status

axelriet, 2021-03-16:
Hi! I'm new to Azure RTOS and just started playing with NXP's MCUExpresso IDE. After installing everything I create a new project using only "Azure RTOS" (no driver or anything) and the project fails to build with the following error:

E:\Nxp\IDE-Workspace\Test\Debug/../azure-rtos/threadx/ports/cortex_m7/mcux/src/tx_initialize_low_level.s:33: **undefined reference to `systick_cycles'**

Please advise,
Thanks!

goldscott, 2021-03-16:
Sounds like NXP messed something up, because all of our ports (https://github.com/azure-rtos/threadx/tree/master/ports/cortex_m7) have SYSTICK_CYCLES defined.

goldscott, 2021-03-19:
Closing this issue. Feel free to re-open if you have more questions.

closed 2021-03-19 by goldscott