eclipse-threadx/threadx #89: _tx_mutex_priority_change preemption doubt

arm7star, 2021-03-18:
https://github.com/azure-rtos/threadx/blob/master/common/src/tx_mutex_priority_change.c line 318~320
                            /* Remember that this thread was preempted by a thread above the thread's threshold.  */
                            TX_MOD32_BIT_SET(next_execute_ptr -> tx_thread_priority, priority_bit)
                            _tx_thread_preempted_maps[MAP_INDEX] =  _tx_thread_preempted_maps[MAP_INDEX] | priority_bit;

why 
"TX_MOD32_BIT_SET(next_execute_ptr -> tx_thread_priority, priority_bit)"
not
"TX_MOD32_BIT_SET(thread_ptr -> tx_thread_priority, priority_bit)"?

I think, in that else case _tx_thread_execute_ptr is equal to next_execute_ptr and next_execute_ptr will be scheduled soon, thread_ptr is preempted.

goldscott, 2021-03-19:
Hi @arm7star - we believe you are correct. Thanks for bringing this up!
How did you find this? Some problem in your code at runtime?

arm7star, 2021-03-23:
I found it unintentionally when i read the codes. Thank you!

closed 2021-03-23 by arm7star

goldscott, 2021-03-23:
@arm7star - we have fixed this and it will be in our next release. Great find and thank you for your help!

goldscott, 2021-04-06:
This has been fixed in https://github.com/azure-rtos/threadx/releases/tag/v6.1.6_rel