eclipse-threadx/threadx #90: _txm_module_manager_thread_create() compile error if   #define  TX_DISABLE_PREEMPTION_THRESHOLD

mapellil, 2021-03-24:
The IAR compilation of file txm_module_manager_thread_create() function _txm_module_manager_thread_create() generates compile error if  #define  TX_DISABLE_PREEMPTION_THRESHOLD  in tx_user.h

Error[Pe020]: identifier "saved_thread_ptr" is undefined C:\AST\AzureRTOS\tmp\prv\AZRTOS_official\Middlewares\ST\threadx\common_modules\module_manager\src\txm_module_manager_thread_create.c 654 

because of:

#ifndef TX_DISABLE_PREEMPTION_THRESHOLD
TX_THREAD               *saved_thread_ptr;
UINT                    saved_threshold =  ((UINT) 0);
#endif

goldscott, 2021-03-25:
I believe we introduced this issue in 6.1.3. We'll fix it. Thanks for bringing this up! I'll leave this ticket open until we release a fix.

andrejm-rtos, 2021-05-24:
Thank you for the report. This has been fixed and it will be part of the next release.

goldscott, 2021-06-25:
This has been fixed here (although the version still says 6.1.3 it is really 6.1.7): https://github.com/azure-rtos/threadx/blob/master/common_modules/module_manager/src/txm_module_manager_thread_create.c

closed 2021-06-25 by goldscott