eclipse-threadx/threadx #94: Potential corruption of ARMv8 NEON and floating-point registers during context save/restore

roxma, 2021-04-09:
I believe there's potential corruption issue of ARMv8 NEON and floating-point registers during context save/restore. 
Let's review the example code in `threadx/ports/cortex_a5x/ac6`.

The irq entry is defined in `example_build/sample_threadx/el3_vectors.S`

```
irqFirstLevelHandler:
  STP      x29, x30, [sp, #-16]!
  BL       _tx_thread_context_save
  BL       irqHandler
  B        _tx_thread_context_restore  
```

Since `_tx_thread_context_save` doesn't save any of NEON and floating-point registers. These registers could be modified in `irqHandler`.

goldscott, 2021-04-09:
Each thread using the FPU hardware needs to call _tx_thread_fp_enable.

roxma, 2021-04-10:
@goldscott Thanks for your reply.

But **_tx_thread_fp_enable DOES NOT HELP**. 

1. The FP registers are actually saved in _tx_thread_context_restore instead of _tx_thread_context_save. It's assuming that **irqHandler** may use FPU hardware. This is a false assumption.
2. It's hard to decide which thread is using FPU hardware. armclang may use `q0~q31` as normal register  even if you don't use float/double or SIMD explicitly. There's no requirement and specification on this.

roxma, 2021-04-12:
@goldscott  To demonstrate my previous point, This example shows how armclang 6.6 uses FP register implicitly.

```
+ cat foo.c
struct foo_t {
    int a0;
    int a1;
    int a2;
    int a3;
};

void foo(struct foo_t *p)
{
    p->a0 = 0;
    p->a1 = 0;
    p->a2 = 0;
    p->a3 = 0;
}

+ armclang --target=aarch64-arm-none-eabi -march=armv8-a -Omax -fno-lto -ffreestanding -c foo.c -o foo.o
+ aarch64-linux-gnu-objdump -D foo.o

foo.o:     file format elf64-littleaarch64

Disassembly of section .text.foo:

0000000000000000 <foo>:
   0:	6f00e400 	movi	v0.2d, #0x0
   4:	3d800000 	str	q0, [x0]
   8:	d65f03c0 	ret
```

goldscott, 2021-04-13:
We originally wrote this with the intent that floating point would not be done in ISRs. I don't think we knew that the compiler would use the VFP registers. 

I suppose you could compile with some option telling the compiler to NOT use VFP registers, or call _tx_thread_fp_enable for every thread.

hwmaier, 2021-08-03:
Does this affect the gcc compiler as well or only clang? And would the `-mgeneral-regs-only` compiler option avoid this issue?

goldscott, 2021-08-05:
This likely affects both clang and gcc.

The ```-mgeneral-regs-only``` flag would alleviate the issue.

closed 2021-08-05 by goldscott