eclipse-threadx/threadx #95: Cortex-a9 gnu example appllication on the Zedboard

ravijj, 2021-04-14:
Hi,

We are working on the cortex_a9 example application build and run it on the Zedboard REV-E.

We are able to build the example application present under the threadx source path:
**"threadx/ports/cortex_a9/gnu/example_build"**

We modified the sample_threadx.ld file to support elf entry point address change to RAM address of the Zedboard.
**@@ -6,12 +6,12 @@ OUTPUT_ARCH(arm)
    __DYNAMIC = 0;    */
 SECTIONS
 {
-  . = 0x00000000;
+  . = 0x1E000000;
     
   .vectors      : {reset.o(.text) }
 
   /* Read-only sections, merged into text segment: */
-  . = 0x00001000;
+  . = 0x1E001000;
   .interp     : { *(.interp)   }**
Refer to the file **sample_threadx.ld** present in the cortex_a9_gnu_sample_threadx.zip file attached below.
With the above changes, we compiled the sample_threax application.

Modified the sample_threax.c to toggle the led in the Zedborad.

1. Toggle led code added in the **threadx_0_entry** and it is not toggling the led, this confirms the threadx_0_entry has some issue in running.
Refer to the file **sample_threadx_thread0_led.c**  present in the cortex_a9_gnu_sample_threadx.zip file attached below.

2. Whereas the same toggle led code is called before the **tx_kernel_enter();** in the **main()** function and it is able to toggle the led.
Refer to the file  **sample_threadx_main_led.c** present in the cortex_a9_gnu_sample_threadx.zip file attached below.

[cortex_a9_gnu_sample_threadx.zip](https://github.com/azure-rtos/threadx/files/6310694/cortex_a9_gnu_sample_threadx.zip)

What could be the issue in calling **tx_kernel_enter();** and thread entries are not running?
Is there anything else that needs to be taken care of, to port this application to work on the Zedbord?

Thanks and Regards,
Ravikiran J J

goldscott, 2021-04-14:
Have you stepped through the code to see if ThreadX is returning any errors?

ravijj, 2021-04-20:
Hi Scott Larson,

We tried debugging the above-reported issue, we found the issue is with the threadX call **tx_thread_sleep(10);** application code execution is crashing at this call.

We are trying CPU private timer enable and periodically calling the **"_tx_timer_interrupt"** in the timer interrupt handler function. we are facing some issues to achieve the same in the Zedboard.

If you have any suggestions regarding the same please let us know.

Thanks and Regards,
Ravikiran J J

goldscott, 2021-04-20:
Please contact azure-rtos-support@microsoft.com and I'll send you an example for Zedboard.

closed 2021-04-21 by goldscott