eclipse-threadx/threadx #98: How to obtain the thread ID?

JesusSmarting, 2021-04-28:
I am doing a port for ThreadX and I need the current thread ID, but I can't find how to get it in ThreadX. 

This function obtains the reference to the actual running thread:

`TX_THREAD      *tx_thread_identify(VOID);`

And with this you obtain the thread info, but there is no ID in the parameters (the most similar is CHAR **name but it is not what I am looking for):

`UINT       tx_thread_info_get(TX_THREAD *thread_ptr, CHAR **name,
              UINT *state, ULONG *run_count, UINT *priority,
              UINT *preemption_threshold, ULONG *time_slice,
              TX_THREAD **next_thread,
              TX_THREAD **next_suspended_thread);`

JesusSmarting, 2021-04-28:
I found current_thread.tx_thread_id might be the solution

goldscott, 2021-04-28:
FYI: tx_thread_id is not a unique number to each thread. It just 0x54485244 for every thread.

JesusSmarting, 2021-04-29:
What should I use then?

goldscott, 2021-04-29:
I'm not sure what you are trying to do. Function tx_thread_identify returns a pointer to the current running thread. You can use this pointer as the "id" ...

closed 2021-04-29 by goldscott

Leeable, 2024-10-30:
Why the tx_thread_id is same for each task? I've created a dozen tasks and each one has a tx_thread_id of 1414025796, I'm a bit confused as to why the tx_thread_id isn't unique.

amgross, 2024-10-31:
tx_thread_id is object type id, it identify the object type. There are different id for thread and mutex and semaphore and so on.

This id purpose is to ensure when getting thread struct pointer that it is really thread struct and not pointer to something else, those object ID's are checked in the txe_XXXXX functions