eclipse-threadx/threadx #102: Running threadx linux SMP port actually on multiple cores

sachindia86, 2021-05-23:
I can see that even though option TX_LINUX_MULTI_CORE is enabled, we are enforcing threadx to run on a single core. With the existing threadx linux SMP port, Is it possible to utilize multiple cores instead of running it on only single core?

TiejunMS, 2021-05-24:
ThreadX for Windows/Linux ports are running as simulators. Each ThreadX thread is 1:1 mapped to native thread. And the native threads are managed by host OS. As a simulator, it does not matter whether native threads are running in single core or multiple cores. If you want to run simulator in multiple cores, try remove the define `-DTX_LINUX_MULTI_CORE` in project (Makefile). Could you give us more details of why you need to utilize multiple cores for simulator?

sachindia86, 2021-05-25:
Thanks for the reply, Tiejun MS
Does that mean threadx scheduler is not scheduling the threads? and underlying OS(win/linux) is taking care of thread scheduling as per its policy.

TiejunMS, 2021-05-25:
For ThreadX Linux port, only one thread is ready at any time. For ThreadX SMP Linux port, the maximum ready threads are limited by simulated core number. It is hard to say underlying OS is doing the scheduling.

sachindia86, 2021-05-25:
Lets assume I have a quadcore machine running linux/windows. Also I have created lets say 4 core bounded threadx threads thread0, thread1, thread1 & thread3 respectively for core0,core1,core3 & core3. Is it guaranteed that these threads will get executed on the assigned cores?

TiejunMS, 2021-05-26:
No it is not guaranteed. What problem are you trying to solve with ThreadX SMP ports for Windows/Linux?

sachindia86, 2021-06-01:
Okay. I just wanted to execute and guanrantee that core bounded tasks should actually execute on the assigned core.

closed 2021-06-15 by sachindia86