eclipse-threadx/threadx #103: Threadx Modules call the same driver problem

HarbourXbr, 2021-05-24:
if i creat A and B Modules need to call the same driver, Will there be a problem of resource competition.
and The compiled volume of AB,will there be two driven volumes.

mransom-campbell, 2021-06-02:
I'm not a ThreadX developer but yes, drivers need to use Mutexes or Semaphores around hardware usage. The ThreadX samples have some examples of this. Here's one: https://github.com/azure-rtos/getting-started/blob/master/Renesas/RSK_RX65N_2MB/app/console.c 

If you're using ThreadX modules then you should probably put device drivers in main application and use the application syscalls to call device functions.

goldscott, 2021-06-09:
Hi @HarbourXbr,

@mransom-campbell is correct.