eclipse-threadx/threadx #107: Modules only map threadx API, other components are not supported

RockySong, 2021-05-31:
Hello, from txm_module.h, only threadx APIs (96 APIs) are defined and mapped to module application code, other components only get there API s reserved, and "txm_module_<component_name>x.h" files are not present, e.g.
txm_module_filex.h
txm_module_netx.h
txm_module_usbx.h
etc
If a module code want to use those components, like filex, guix, how to invoke their APIs from resident code?

mransom-campbell, 2021-06-02:
I was wondering the same thing. I'm not sure if the APIs are just reserved for future implementation or if they have an implementation that isn't on GitHub.

GetFreedomPro, 2021-06-03:
Looking forward to later official to improve...

goldscott, 2021-06-09:
Hi all, 

Official answer: FileX, NetX, USBX, etc have not yet been ported to Modules. We are working on it.

closed 2021-06-09 by goldscott